atomic14/esp32_sdcard_audio #1: SD Card Wiring with pull-up Resistors and code issue

dtiziano, 2021-04-26:
I'm trying to run this project.

First
When trying to write to the SD Card using the idf code, it always fails to initialize the SD card. It is as HC card though.
Did you use any pull-up resistors? If yes, can you provide your connection scheme?

Furthermore, I would like to use the internal ADC.
After commenting //#define USE_I2S_MIC_INPUT in config.h

there will be two bugs in line 97 of main.cpp
97: I2SSampler *m_input = new ADCSampler(ADC_UNIT_1, ADC1_CHANNEL_7);

1.  shouldn't *m_input be *input?
2. ADCSampler requires an additional parameter, is this i2s_adc_config?

I would propose the solution  following solution, for which I'm still not sure since I cannot test it as long as I can have the SD card working

  I2SSampler *input = new ADCSampler(ADC_UNIT_1, ADC1_CHANNEL_7, i2s_adc_config);

rmond18, 2021-05-10:
I tried the Arduino code with I2S microphone and I also have a similar error:

`E (628) SDC: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
`

Can you, @cgreening, please share your wiring diagram if possible. It would be a great help. 
Thank you very much!

Warm regards,
rmond

rmond18, 2021-05-10:
I have one last question, can the speaker be replaced by an I2S stereo decoder such as UDA1334A for example? or not?

Thank you for replying! I'll review my pinout to see what's wrong, thank you!

dtiziano, 2021-05-11:
In the meantime I've got it working. For me as well, I'm not using any pull up resistors

1. Format the SD card using the Official SD Card Formatter
2. Check that the SD card Pins are connected to the defined SD card GPIO pins defined in config.h (they're not the standard SPI pins)
3. In main.cpp, 

change according to the already fix commit https://github.com/atomic14/esp32_sdcard_audio/commit/7998e3f27c46bbb4af6bd1388d32ec43d3948ab2#diff-1bf9bda525f4a5ffea55672840f8c89cbcf45ab885a7e5937b53fde9be1c332b

cgreening, 2021-05-11:
That's great to hear!

closed 2021-05-20 by cgreening

Yatinbaluja, 2021-05-21:
@cgreening I don't want to use the push button to record. what needs to be done.

cgreening, 2021-06-14:
Can you create a new issues for your problem.