atomic14/esp32_sdcard_audio #4: converting data from int32_t to int16_t

dtiziano, 2021-06-12:
Hi Atomic14
I've got a question about what is happening in line 35 of https://github.com/atomic14/esp32_sdcard_audio/blob/main/idf-wav-sdcard/lib/audio_input/src/I2SMEMSSampler.cpp where the following code is executed. 

samples[i] = (raw_samples[i] & 0xFFFFFFF0) >> 11; 

where raw_samples is a int32_t, which is filled with values that can be as high as the microphone resolution, e.g. 24 bit for the ICS43434. 
Am I right, assuming that operating &  0xFFFFFFF0 on values values confined to 24bit results in unchanged values, and is thus a useless operation? I'm then not sure why the left shift of 11 bits? Here we would like to end having "samples" of the size of 16bit, thus I don't understand why shifting right by 11, shouldn't this be either >> 32-16=16 or 32-24=8?

Thanks a lot

Tiziano

cgreening, 2021-06-12:
Hi, yes, those couple of lines deserve some comments in the code!

You are correct with the current code the `& 0xFFFFFFF0` will do nothing as we're shifting the lower bits out anyway.

I kind of left it in there in case I wanted to switch to the full 24 bits and wanted to make sure the lower 8 bits were definitely cleared so there was no noise.

The shift of 11 bits is simply because of the very high dynamic range of the mic - I found that shifting by 16bits would only keep very loud sounds and would lose quiet sounds. So in the end I settled on shifting by 11 bits. You will get some pretty horrible noise if you are recording very loud sounds.

Ideally, it would be nice to have some automatic gain control or compression in the conversion from 24 bit to 16 bit. Or even more ideally, capture the samples at the full 24 bits and only monkey around with the volume on the playback side of things. But then you are doubling the storage requirements which is probably fine for SDCard recording, but not so good for recording to SPIFFS.

I'll add some comments to the code to explain the thinking behind the code as it's not very obvious!

dtiziano, 2021-06-14:
Comments would be great 

Do you know  if it would be possible to record the data in its original 24bit? Is that allowed for the WAV file format? 

Basically, converting `write(int16_t *samples, int count)` into `write(int32_t *samples, int count)`

cgreening, 2021-06-14:
Yes definitely possible. You'll have to modify all the `int16_t`  to `int32_t` and also modify the contents of `lib/wav_file/src/WAVFile.h`.

dtiziano, 2021-06-14:
great, thx for the hint.

I still don't get a thing. By using your original code, the data is >>11. I'm using a ICS43434 which provides data in 24bit format. Thus the max value i should record when the microphone saturates should be 2^23-1 >>11 ~= 2^11-1 but when I open the wav file on the computer everything seems to be scaled to have maximum amplitude of 1 (data is normed). Where does the norm process happen?

closed 2021-06-16 by cgreening