atomic14/esp32_sdcard_audio #6: BT sink integration + failed sd card initialization

rmond18, 2021-06-14:
Hi,

1. Is it possible to integrate the recording feature to a BT sink? 
I'm currently working on integrating this to a [BT sink](https://github.com/tierneytim/btAudio), where instead of the I2S mic, the audio data will be coming from a BT source. 

However I'm still having a hard time on making your code work on my esp32 board, I still got problems with initialization of the sd card. 
[image]
I'm using a microSD card reader module with SanDisk Ultra (16GB) MicroSD HCI formatted as FAT32. 
2. Do you have any idea/tips on how to debug this? 

I tried using the SD card with [other](https://github.com/MhageGH/esp32_SoundRecorder) sound recording code and it worked, still puzzled on why I can't initialize the SD card using your code.

Warm regards,
rmond

cgreening, 2021-06-15:
Have you updated `config.h` to match the pins you are using for your SDCard?

```
// sdcard
#define PIN_NUM_MISO GPIO_NUM_4
#define PIN_NUM_CLK GPIO_NUM_14
#define PIN_NUM_MOSI GPIO_NUM_15
#define PIN_NUM_CS GPIO_NUM_25
```

rmond18, 2021-06-15:
Yes, I have updated config.h to match the pins used in my SD card.

cgreening, 2021-06-15:
I think the problem is that you've got pins used for the SD card that are also used for the I2S amplifier.

```
// speaker settings
#define I2S_SPEAKER_SERIAL_CLOCK GPIO_NUM_19
#define I2S_SPEAKER_LEFT_RIGHT_CLOCK GPIO_NUM_27
#define I2S_SPEAKER_SERIAL_DATA GPIO_NUM_18
```

You'll need to move the I2S_SPEAKER_SERIAL_CLOCK and I2S_SPEAKER_SERIAL_DATA to different pins.

rmond18, 2021-06-15:
I also changed them beforehand and made sure that no pins are used more than once.  I'm actually using UDA1334A decoder instead of an I2S amplifier.
[image]

Is there a difference in using a breakout microsd card adapter (what I'm using) and using a sd card adapter with soldered pins (the one that you used in the video)? I'm thinking if my breakout board is the culprit of the failed sd initialization but it works fine on other code.

cgreening, 2021-06-15:
It's possible, but I don't see why there would be any difference. SD Cards do seem to be very finickity. There's no reason why you couldn't modify my code to use the standard SD.h which is what is being used in the example you linked to.

Or I would just use the code that works as it looks like it does a very similar thing to this repo.

rmond18, 2021-06-15:
I tried soldering pins to an sd adapter and used it, finally the SD card worked.
[image]

However, the .wav file doesn't have any sound, now I'll try to debug this. Thank you for answering my questions!

cgreening, 2021-06-16:
That's interesting - what was the SD card adaptor you were using?

To answer your other question - that's a bit outside the scope of this demo repository. I would suggest reading through the current code to see how samples are written to the SD Card. If you have a callback with a set of samples then you should be able to take it from there.

closed 2021-06-16 by cgreening

rmond18, 2021-06-16:
Here is the SD adapter that I soldered that worked:
[image]

Here is the breakout board that didn't worked:
[image]

Some level shifting SD card adapters won't work according to [this](https://www.reddit.com/r/esp32/comments/d71es9/a_breakdown_of_my_experience_trying_to_talk_to_an/).
I'll study your code more on how you write to the SD card, thank you very much again!