atomic14/esp32_sdcard_audio #7: SDCard mount error;

7doctor7, 2021-06-16:
Hi!
I tried to use your IDF version code and can't mount sd card

I (343) app: Starting up
I (343) app: Mounting SDCard on /sdcard
I (353) SDC: Initializing SD card
I (353) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (403) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (403) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (403) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (413) SDC: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.
I (423) app: Creating microphone

I've redefined sd card pins on config.h 

#define PIN_NUM_MISO GPIO_NUM_19
#define PIN_NUM_MOSI GPIO_NUM_23
#define PIN_NUM_CLK GPIO_NUM_18
#define PIN_NUM_CS GPIO_NUM_5

and temporarily disable output logic and speaker pins...

maybe you can provide some help?
thanks

EDsteve, 2021-06-20:
I had similar issues and for me it helped to pull up the SD card pins via code in main.cpp:
  gpio_set_pull_mode(PIN_NUM_MOSI, GPIO_PULLUP_ONLY); 
  gpio_set_pull_mode(PIN_NUM_CLK, GPIO_PULLUP_ONLY);
  gpio_set_pull_mode(PIN_NUM_MISO, GPIO_PULLUP_ONLY);
  gpio_set_pull_mode(PIN_NUM_CS, GPIO_PULLUP_ONLY);

I add it here and it works, but i am no coder. So please advice if there is a better place for it: 
`  while (true)
  {
    // wait for the user to push and hold the button
    wait_for_button_push();
  gpio_set_pull_mode(PIN_NUM_MOSI, GPIO_PULLUP_ONLY); 
  gpio_set_pull_mode(PIN_NUM_CLK, GPIO_PULLUP_ONLY);
  gpio_set_pull_mode(PIN_NUM_MISO, GPIO_PULLUP_ONLY);
  gpio_set_pull_mode(PIN_NUM_CS, GPIO_PULLUP_ONLY);`

cgreening, 2021-06-20:
Probably the best place would be in the SDCard initialisation code:

Line 19 of `lib/sd_card/src/SDCard.cpp`

```
SDCard::SDCard(const char *mount_point, gpio_num_t miso, gpio_num_t mosi, gpio_num_t clk, gpio_num_t cs){
```

But I'm wondering if external pull-ups would be better. I'll add a note to the `README.md` file about this.

cgreening, 2021-06-20:
30f087fcf7d094cd20b038bbf94332e9176a08d7

closed 2021-06-20 by cgreening