atomic14/esp32_sdcard_audio #9: Using Arduino SD library;

7doctor7, 2021-06-26:
Hi!
Maybe you can provide some help for using the Arduino-provided SD library for saving WAV files to SD card?
Because now I can't do it with stdio.h in yours WAVFileWriter and Arduino SD and FS libs.
Thank in advance.

beustens, 2022-11-27:
First, change the contents of the following files:

**WAVFileWriter.h**

```c++
#pragma once

#include "FS.h"
#include "WAVFile.h"

class WAVFileWriter
{
private:
  int m_file_size;

  File m_fp;
  wav_header_t m_header;

public:
  WAVFileWriter(File fp, int sample_rate);
  void start();
  void write(int16_t *samples, int count);
  void finish();
};
```

**WAVFileWriter.cpp**

```c++
#include "WAVFileWriter.h"

static const char *TAG = "WAV";

WAVFileWriter::WAVFileWriter(File fp, int sample_rate)
{
  m_fp = fp;
  m_header.sample_rate = sample_rate;
  // write out the header - we'll fill in some of the blanks later
  m_fp.write((uint8_t *) &m_header, sizeof(m_header));
  m_file_size = sizeof(wav_header_t);
}

void WAVFileWriter::write(int16_t *samples, int count)
{
  // write the samples and keep track of the file size so far
  uint8_t* sampleBytes = (uint8_t*)samples;
  m_fp.write(sampleBytes, sizeof(int16_t) * count);
  m_file_size += sizeof(int16_t) * count;
}

void WAVFileWriter::finish()
{
  // now fill in the header with the correct information and write it again
  m_header.data_bytes = m_file_size - sizeof(wav_header_t);
  m_header.wav_size = m_file_size - 8;
  m_fp.seek(0);
  m_fp.write((uint8_t *) &m_header, sizeof(m_header));
}
```

Then, in **main.cpp**, you can change the `record` function:

```c++
#include "SD.h"
#include "FS.h"
#include <WAVFileWriter.h>

#define MIC_BUFFER_LEN 1024
#define MIC_SAMPLERATE 16000

void record(const char *fname) {
	int16_t *samples = (int16_t *)malloc(sizeof(int16_t) * MIC_BUFFER_LEN);
	startMic();
	// open the file on the sdcard
	File fp = SD.open(fname, FILE_WRITE);
	// create wav file writer
	WAVFileWriter *writer = new WAVFileWriter(fp, MIC_SAMPLERATE);
	// keep writing until the user releases the button
	while (gpio_get_level(BTN_REC) == 1) {
		int samples_read = readMic(samples, MIC_BUFFER_LEN);
		writer->write(samples, samples_read);
	}
	// stop the input
	stopMic();
	// and finish the writing
	writer->finish();
	fp.close();
	delete writer;
	free(samples);
}
```

cgreening, 2023-05-27:
@beustens I know this is an old issue - but if you want to turn your changes into a PR I'd be happy to merge them in.