atomic14/esp32_sdcard_audio #15: E (17) vfs_fat_sdmmc: slot init failed (0x103).

waqas-subhani, 2022-11-03:
Hi, 
I am using TTGO-T1, during compliation no error , after upload this error ocurr, previously worked fine, I have Change nothing in code. Now this appears

E (17) vfs_fat_sdmmc: slot init failed (0x103).
[    23][E][SDCard.cpp:66] SDCard(): [SDC] Failed to initialize the card (ESP_ERR_INVALID_STATE).

scottky00, 2022-11-07:
I am getting this same error

Sunlu2333, 2022-11-15:
> I am getting this same error

I have encountered the same problem. Can you share with me if you have found the solution？

7doctor7, 2022-11-15:
Look into https://github.com/atomic14/esp32_sdcard_audio/issues/10 issue...

waqas-subhani, 2022-11-21:
Hi, I have solved this error by adding lines and commet  the  lines in the code. 

 // ESP_LOGI(TAG, "Initializing SD card");
 // Set up a configuration to the SD host interface
  sdmmc_host_t host_config = SDSPI_HOST_DEFAULT();
  /* These Lines is deprecated in new updates
  sdspi_slot_config_t slot_config =SDSPI_SLOT_CONFIG_DEFAULT();
  slot_config.gpio_miso = miso;
  slot_config.gpio_mosi = mosi;
  slot_config.gpio_sck = clk;
  slot_config.gpio_cs = cs;*/

      // Set up SPI bus
  spi_bus_config_t bus_cfg =
      {
          .mosi_io_num = mosi,
          .miso_io_num = miso,
          .sclk_io_num = clk,
          .quadwp_io_num = -1,
          .quadhd_io_num = -1,
          .max_transfer_sz = 4000};
          
  spi_bus_initialize(HSPI_HOST, &bus_cfg, 1);
  sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
  slot_config.gpio_cs = cs;
  slot_config.host_id = SPI2_HOST;

  /// ret = esp_vfs_fat_sdmmc_mount(m_mount_point.c_str(), &host_config, &slot_config, &mount_config, &m_card);  Line is deprecated
  ret = esp_vfs_fat_sdspi_mount(m_mount_point.c_str(), &host_config, &slot_config, &mount_config, &m_card);

closed 2022-11-21 by waqas-subhani