atomic14/esp32_sdcard_audio #16: FileRead using Arduino's SD Library

tuneado, 2023-03-14:
Hey There!
Thanks alot for this awsome project! its been very helpful and made my life so much easier for my voice memo project

I'm migrating the project to use Arduino's SD library because i am not master at coding and im more used to use the SD Library.
So far so good i managed to record the audio and do some more stuff that i wanted.
But im completely lost when it come to adapt the FileRead code.

Any Help would be appreciated
Best Regards

7doctor7, 2023-03-15:
Hi
try https://github.com/atomic14/esp32_sdcard_audio/issues/9#issuecomment-1328352905
regards

tuneado, 2023-03-15:
Hey
Already had a look into #9 before i poster this issue. I indeed managed to get the write functions to work but i couldnt with the Read functions

tuneado, 2023-03-16:
Ok i managed to get it to work..with some issues tho
Im writing at 1024 samples and reading at 2048...what am i doing wrong?
here's the code
**WAVFileReader.cpp**
`#include "esp_log.h"
#include "WAVFileReader.h"

static const char *TAG = "WAV";

WAVFileReader::WAVFileReader(File fp)
{
    m_fp = fp;
    // read the WAV header
    // fread((void *)&m_header, sizeof(wav_header_t), 1, m_fp);
    m_fp.read((uint8_t *)&m_header, sizeof(m_header));
    // sanity check the bit depth
    if (m_header.bit_depth != 16)
    {
        ESP_LOGE(TAG, "ERROR: bit depth %d is not supported\n", m_header.bit_depth);
    }
    if (m_header.num_channels != 1)
    {
        ESP_LOGE(TAG, "ERROR: channels %d is not supported\n", m_header.num_channels);
    }
    ESP_LOGI(TAG, "fmt_chunk_size=%d, audio_format=%d, num_channels=%d, sample_rate=%d, sample_alignment=%d, bit_depth=%d, data_bytes=%d\n",
             m_header.fmt_chunk_size, m_header.audio_format, m_header.num_channels, m_header.sample_rate, m_header.sample_alignment, m_header.bit_depth, m_header.data_bytes);
}

int WAVFileReader::read(int16_t *samples, int count)
{
    uint8_t* sampleBytes = (uint8_t*)samples;
    size_t read =m_fp.read(sampleBytes, sizeof(int16_t) * count); // fread(samples, sizeof(int16_t), count, m_fp);
    return read;
}`

**WAVFileReader.h**

`#pragma once
#include "FS.h"
#include "WAVFile.h"
//#include <stdio.h>

class WAVFileReader
{
    private:
        File m_fp;
        wav_header_t m_header;

    public:
        WAVFileReader(File fp);
        int sample_rate() { return m_header.sample_rate; }
        int read(int16_t *samples, int count);
};`

And changes made on the  play function

`void play(Output *output, const char *fname)
{
  int16_t *samples = (int16_t *)malloc(sizeof(int16_t) * 1024);
  // open the file on the sdcard
  fp = SD.open(fname , FILE_READ);
  // create a new wave file writer
  WAVFileReader *reader = new WAVFileReader(fp);
  ESP_LOGI(TAG, "Start playing");
  output->start(reader->sample_rate());
  ESP_LOGI(TAG, "Opened wav file");
  // read until theres no more samples
  while (true)
  {
    int samples_read = reader->read(samples, 1024);
    if (samples_read == 0)
    {
      break;
    }
    ESP_LOGI(TAG, "Read %d samples", samples_read);
    output->write(samples, samples_read);
    ESP_LOGI(TAG, "Played samples");
  }
  // stop the input
  output->stop();
  fp.close();
  delete reader;
  free(samples);
  ESP_LOGI(TAG, "Finished playing");
}`

closed 2023-04-05 by tuneado