atomic14/esp32_sdcard_audio #18: Errors when building project in Platformio using Project's IDF code files

that-one-arab, 2023-05-26:
## Issue
When using the project's IDF version, I get the following error when building the project using Platformio:
```
src/main.cpp:45:21: error: 'esp_timer_get_time' was not declared in this scope; did you mean 'timer_gettime'?
```

## What I tried
Resolved the error by including the `esp_timer.h` header in `main.cpp`
```
#include "esp_timer.h"
```

Doing that gets rid of the above error but gives me a new error:
```
lib/audio_output/src/I2SOutput.cpp:13:24: error: narrowing conversion of 'sample_rate' from 'int' to 'uint32_t' {aka 'long unsigned int'} [-Werror=narrowing]
lib/audio_output/src/DACOutput.cpp:9:24: error: narrowing conversion of 'sample_rate' from 'int' to 'uint32_t' {aka 'long unsigned int'} [-Werror=narrowing]
```
I now believe that this is a GCC version errors, from the research I did it appears that GCC 4.8.0 ignores these conversion errors.
I thought that I could manually patch fix these lines of codes but that gives me even more errors.

I tried to downgrade `toolchain-xtensa-esp32`
```platformio.ini
[env:esp32dev]
platform = espressif32
platform_packages =
  toolchain-xtensa-esp32@11.2.0+2022r1
board = esp32dev
framework = espidf
upload_port = /dev/cu.SLAB_USBtoUART
monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_build.partitions = partitions.csv
```

Which is the package that contains the GCC compiler used for the project, but older versions of this package gives me CLI errors

## Environment
Windows 10
PACKAGES:
 - framework-espidf @ 3.50001.0 (5.0.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1

Can anyone please help me? I really need to compile this project for my university 😓

michaelbrown1, 2023-05-26:
Try changing `.sample_rate = sample_rate` to `.sample_rate = (uint32_t)sample_rate,` in the DACOutput.cpp and I2SOutput.cpp files

that-one-arab, 2023-05-26:
@michaelbrown1 The project compiled! Thank you!

that-one-arab, 2023-05-26:
@michaelbrown1 Would it be worthwhile for me to release a PR for this fix?

closed 2023-05-26 by that-one-arab

cgreening, 2023-05-27:
Yes!

that-one-arab, 2023-05-27:
@cgreening Done. I am a huge fan of your channel! Thank you so much for your video guides and contributions to open source!