atomic14/esp32_sdcard_audio #20: Analog microphone slow and noisy recordings

that-one-arab, 2023-06-01:
## Issue
When recording with an analog microphone (MAX9814) the recordings are low in pitch, slow and noisy. When I manually increase the sample rate using software like Audacity I get significantly closer to the original recording (although sound pitch is off) but it is still very noisy.

## Environment
Using IDF project
Windows 10
ESP32 development board
MAX 9814

Beside merging changes inside this [PR](https://github.com/atomic14/esp32_sdcard_audio/pull/19) no other code change was made. Not even to the config (I stuck by the original pinout)

I am unsure how to begin debugging this.

tuneado, 2023-06-28:
i'm having axactly the same issue. I've opened an issue thread regardig this.
Are you also using the warduino SDCard Library?

valentasn, 2023-08-24:
I had some issues to, maybe this will help:
https://github.com/atomic14/esp32_sdcard_audio/issues/17#issuecomment-1692335765